EverythingSmartHome/everything-presence-lite #248: Can't connect to Wifi

achoke6, 2024-11-27:
Just got my Everything Presence Lite. Home assistant found the bluetooth/improv listing, but wouldn't connect to my wifi network. I connected, flashed firmware and can't connect to wifi for the life of me. I tried the bluetooth and non-bluetooth versions.
[esp-web-tools-logs (3).txt](https://github.com/user-attachments/files/17939314/esp-web-tools-logs.3.txt)

EverythingSmartHome, 2024-11-27:
First thing to try would be a different power supply, WiFi issues are often a result of bad power delivery

achoke6, 2024-11-29:
that worked thanks! Now i just need to figure out why my presence is always
shown at the top of the radar.

unlucio, 2024-12-30:
I'm having the same issue, I successfully configured 3 EPL ed a EPOne, I'm now adding the last EPL and it's making bang my head to the wall as it refuses to connect to my wifi network.
I tried different power supply, switched to one of the already working one, even a 140W GaN one, disconnected several devices form my wifi just in case, but nothing.
I keep getting:
- Enter Wi-Fi credentials to connect the device to your network.
- enter my wifi credential, and I'm 100% positive I haven't forgot them 😂
- triple check the wifi info are correct, I even went into Keychain and confirmed both SSID and password
- submit
- the spinner spins
- I get: "The device is connecting to the Wi-Fi network." with a spinner
- and boom, back to the beginning with the following error: The device could not connect to the Wi-Fi network. Check that the SSID and password are correct and try again.

I tried rebooting pretty much everything, power cycled the unruly EPL, but nothing. I'm stuck in that silly loop :/
It worked for 4 other, can it be I just got a bad one?

MrBeats, 2025-01-12:
Same problem here

henrikhjelmse, 2025-03-29:
same here. cant connect to wifi. same problem.

unlucio, 2025-03-30:
@MrBeats @henrikhjelmse I was suggest by @EverythingSmartHome to reset my sensor

 https://everythingsmarthome.github.io/everything-presence-lite/updating.html#manual-setup-and-factory-resetting

And it worked for me.

henrikhjelmse, 2025-03-30:
did try that and flash new firmware.
did not work.