EverythingSmartHome/everything-presence-lite #251: What’s the latest Firmware, on 1.0.3 but cant see updates and getting errors with Zone Configurator

For006, 2024-11-28:
Is 1.0.3 the latest, as cant see some entities such as the new angle one for zone configurator?

EverythingSmartHome, 2024-11-28:
1.2.0 is the latest

For006, 2024-11-29:
I’ve tries several times to update, but keep getting errors. Restarted HA etc.

ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
 [ERROR] /config/esphome/everything-presence-lite-24e970.yaml 
 
 Any ideas?

EverythingSmartHome, 2024-11-29:
Can you post the full config and full logs?

For006, 2024-12-07:
Hi, 

HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
RAM:   [=         ]  13.7% (used 45024 bytes from 327680 bytes)
Flash: [========  ]  84.7% (used 1553360 bytes from 1835008 bytes)
========================= [SUCCESS] Took 46.47 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of everything-presence-lite-24e970.local in mDNS
INFO Connecting to 192.168.0.2 port 3232...
INFO Connected to 192.168.0.2
INFO Uploading /data/build/everything-presence-lite-24e970/.pioenvs/everything-presence-lite-24e970/firmware.bin (1558800 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 15.05 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
 [ERROR] /config/esphome/everything-presence-lite-24e970.yaml 

======================== [SUMMARY] ========================
  - /config/esphome/everything-presence-lite-24e970.yaml: FAILED

For006, 2024-12-07:
I've tried connecting to USB and using the HLKRadarTool App on iOS which did seem to update.
But can't get it updated using Home Assistant - and not sure how to find the MQQT log.

Any suggestions?

For006, 2024-12-07:
Updating /config/esphome/everything-presence-lite-24e970.yaml
------------------------------------------------------------

INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/everything-presence-lite-24e970.yaml...
WARNING GPIO8 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-lite-24e970 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
RAM:   [=         ]  13.7% (used 45024 bytes from 327680 bytes)
Flash: [========  ]  84.7% (used 1553360 bytes from 1835008 bytes)
========================= [SUCCESS] Took 46.47 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of everything-presence-lite-24e970.local in mDNS
INFO Connecting to 192.168.0.2 port 3232...
INFO Connected to 192.168.0.2
INFO Uploading /data/build/everything-presence-lite-24e970/.pioenvs/everything-presence-lite-24e970/firmware.bin (1558800 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 15.05 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
 [ERROR] /config/esphome/everything-presence-lite-24e970.yaml 

======================== [SUMMARY] ========================
  - /config/esphome/everything-presence-lite-24e970.yaml: FAILED

For006, 2024-12-07:
@EverythingSmartHome 
Added the code, thanks.

EverythingSmartHome, 2024-12-08:
Need to see the contents of the config file

EverythingSmartHome, 2024-12-08:
There also doesnt look like an issue here from the logs as it says 100% upload complete (I know there is an error after but could be a red herring as the upload process completed OK)

For006, 2024-12-09:
> Need to see the contents of the config file

Sorry, not sure how to do that.
Is there a way to force update via PC and USB?

For006, 2024-12-09:
@EverythingSmartHome  Think it's sorted, 
on your site, Updating and Connecting To WiFi, but Connect was not showing.
clicked select smart home platform, and got it installed and updated that way.

Took some figuring out!

Thanks for responding, now it's updated will try the mapping.

EverythingSmartHome, 2024-12-09:
The connect button only shows up after you make the selection for the options you would like as you mentioned, that's intentional (so that the correct firmware gets installed depending on what you need)

Glad it's working!

closed 2024-12-09 by EverythingSmartHome