EverythingSmartHome/everything-presence-lite #257: Everything Presence Zone Configurator - error

MR-Apple-uk, 2024-11-30:
Everything Presence Zone Configurator - Status: Error Fetching Data

Hi, the zone Configurator isnt working i just get error fetching data please see my logs below 

s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a Home Assistant Add-on.
 * Serving Flask app 'backend'
 * Debug mode: off

EverythingSmartHome, 2024-11-30:
Make sure you have all entities enabled in Home Assistant, some are hidden by default

jwall87, 2024-11-30:
Same problem here. Tried enabling all entities but still persistent.

jwall87, 2024-11-30:
> > Make sure you have all entities enabled in Home Assistant, some are hidden by default
> 
> Thanks worked for me after enabling:)

Thanks for the response.

Unfortunately, after verifying and restarting I'm getting the same error messages:

[image]

EverythingSmartHome, 2024-11-30:
This looks like an EP1? The zone configurator is for the Lite as the EP1 doesn't support zones

closed 2024-11-30 by EverythingSmartHome

jwall87, 2024-11-30:
> This looks like an EP1? The zone configurator is for the Lite as the EP1 doesn't support zones

Ok that would be why. Thank you, I didn't realize that.

EverythingSmartHome, 2024-11-30:
No worries!