EverythingSmartHome/everything-presence-lite #259: Add Humidity and Temperature support for SCD40 sensor

hajar97, 2024-12-03:
Currently, SCD40 sensor exposes only CO2 information. It is definitely capable of reading humidity and temperature information as well. I wonder if there is any specific reason why it is not being currently added? For example, you have been testing its addition and came to conclusion that its readings are insufficiently reliable and hence you prefer not to expose them? If so, it would be great for the community to know. 

Until you share that information, here is a simple code to add to ESPHome YAML configuration to expose readings from those sensors including customisable offsets that may be necessary in some scenarios. If there is no objective reason for not exposing readings from these sensors it would be great if you could add their support officially. Thank you.

```
sensor:
  - platform: scd4x
    i2c_id: bus_b
    co2:
      name: "CO2"
    temperature:
      name: "Temperature"
      filters:
        - lambda: return x - id(temperature_offset_ui).state;
    humidity:
      name: "Humidity"
      filters:
        - lambda: return x - id(humidity_offset_ui).state;
    automatic_self_calibration: false

number:
  - platform: template
    name: "Temperature Offset"
    id: temperature_offset_ui
    unit_of_measurement: "°C"
    min_value: -70
    max_value: 70
    step: 0.1
    mode: box
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 15.0
    icon: "mdi:temperature-celsius"
    entity_category: config
  
  - platform: template
    name: "Humidity Offset"
    id: humidity_offset_ui
    unit_of_measurement: "%"
    min_value: -70
    max_value: 70
    step: 0.1
    mode: box
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: -20.0
    icon: "mdi:cloud-percent"
    entity_category: config
```

EverythingSmartHome, 2024-12-03:
We intentionally do not expose (or even advertise temperature and humidity in the shop description) because they are really not accurate at all from our testing (and testing done by others too), and so we prefer to just focus on the CO2 side. Users are of course free to add temp and humidity options through their own ESPHome if they wish but it's not something we would add at this time.

closed 2024-12-03 by EverythingSmartHome

neilmartin83, 2024-12-03:
I've found the humidity sensor to be wildly off as the higher internal ambient temps of the unit will definitely affect the amount of moisture in that little pocket of air. Even an offset will likely not help much because air holds different levels of water vapour based on temperature.

However the temperature sensor is surprisingly accurate with the 15 degree offset - I'm getting +/- 1 degree compared to other sensors in the same two rooms I have these in which is good enough for a quick glance/dashboard type application.

hajar97, 2024-12-03:
I wonder if you have done your tests with sensor cap open or closed? Surely, if we want to use CO2, Temperature and Humidity sensors, then it makes sense to make a number of holes in the frontal cap of the device, to make sure air can go in/out more easily. Wonder if I do that would it make the humidity readings more reliable?

hajar97, 2024-12-03:
According to sensor's manufacturer it has 0.8 °C temperature accuracy and 6% relative humidity accuracy, which if true is actually quite accurate for a simple indoor use.

https://sensirion.com/products/catalog/SCD40

hajar97, 2024-12-05:
I have noticed that opening or closing the frontal cap makes a big difference to the readings and changes are observable within 10 mins. As an example, having the cap off and with readings of Humidity 68.62% and Temperature 14.88, after putting the cap on within 10 mins readings changed to Humidity being 58.22% and Temperature 22.66. It probably shows how heat from the PCB makes the air much dryer and temperature much higher as soon as cap is on and outside air doesn't flow quite as easily.