EverythingSmartHome/everything-presence-lite #260: Unable to add everything presence lite -

williamjeccles, 2024-12-03:
So when plugging in the device to power, it appears within home assistant automatically.

When I click Add, I see the following spinner for a while
[image]

Then the following error:
[image]

Any ideas on what to do next?

EverythingSmartHome, 2024-12-03:
Try restarting Home Assistant and try again

williamjeccles, 2024-12-03:
done, no difference

EverythingSmartHome, 2024-12-03:
Hmm strange, that bit is handled by Bluetooth on Home Assistant.

In that case I would try the manual setup process in the user guide, below the automatic setup section

williamjeccles, 2024-12-11:
Updating the firmware and connecting via the manual method worked for me. Thanks

closed 2024-12-11 by williamjeccles