EverythingSmartHome/everything-presence-lite #262: Zone Configurator: One EPL works, other gets `Status: Error Fetching Data`

caretakerscurse, 2024-12-05:
I have two EPLs. One that's been in use since receiving, and one that was adopted, but left unused, until today.

I plugged in the second one, and went the the configurator, but noticed that only EPL-01 would load, EPL-02 gets `Status: Error Fetching Data`

Both are latest FW. Although EPL-01 was working fine without everything enabled. They both have everything enabled now, still EPL-02 gets the error.

I tried to find how to enabled debug logs, but couldn't find a toggle nor instructions.

I've restarted HA, reinstalled EPZC, and probably a few other things, but without a debug log, I'm just taking stabs in the dark. 

Any idea what's going on?

Thank you <3

############

Everything Presence Zone Configurator
```
Current version: 1.1.0
```
EPL-01
```
Firmware: 1.2.0 (ESPHome 2024.11.0)
Hardware: 1.0.3
MAC: 30:C9:22:F1:B4:30 
```
EPL-02
```
Firmware: 1.2.0 (ESPHome 2024.11.0)
Hardware: 1.0.3
MAC: FC:E8:C0:4E:04:08 
```

EPZC Log:
```
...
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a Home Assistant Add-on.
 * Serving Flask app 'backend'
 * Debug mode: off
 ```

EverythingSmartHome, 2024-12-05:
You need to enable all the zones in Home Assistant, zones 2-4 are hidden by default. Enable these and should be good

caretakerscurse, 2024-12-05:
> They both have everything enabled now, still EPL-02 gets the error.

Sorry I wasn't clear, all **entities** are enabled.

[image]

EverythingSmartHome, 2024-12-05:
Do you see any issue or is it just the error? Is it showing but works OK?

caretakerscurse, 2024-12-05:
The configurator is unusable for EPL-02, but otherwise the device appears to work fine. Tracks objects, etc.

EverythingSmartHome, 2024-12-06:
Sorry I meant the configurator. What happens? No info displayed at all or some or something else?

caretakerscurse, 2024-12-06:
For 02, the only change is the radius grid image loads. The target table does not load any data. 

For 01, the table grid work as expected

Init
[image]
EPL-02
[image]
EPL-01
[image]

EverythingSmartHome, 2024-12-06:
Have you changed any of the names of the entities per chance? It might be struggling to auto detect the entities if the name has changed

caretakerscurse, 2024-12-06:
only the name of the device, (EPL-02) same as I did with the first (EPL-01).

I even tried renaming the device to rename all the entities without any progress.

caretakerscurse, 2024-12-06:
I did just check what differs from 01 and 02 in renaming, and everything is the same on both except for a unifi entity and one called `sensor.uptime_64` 

01:
[image]

02:
[image]

EverythingSmartHome, 2024-12-09:
What about the other 56 entities?

caretakerscurse, 2024-12-09:
They are named as expected, as far as I can tell.

(again using the rename tool, only for the sake of getting the entity names)

EPL-01:
```
Old 	New
button.epl_01_esp_reboot 	button.epl_01a_esp_reboot
light.epl_01_esp32_led 	light.epl_01a_esp32_led
sensor.epl_01_illuminance 	sensor.epl_01a_illuminance
number.epl_01_illuminance_offset 	number.epl_01a_illuminance_offset
number.epl_01_max_distance 	number.epl_01a_max_distance
binary_sensor.epl_01_occupancy 	binary_sensor.epl_01a_occupancy
number.epl_01_occupancy_off_delay 	number.epl_01a_occupancy_off_delay
binary_sensor.epl_01_target_1_active 	binary_sensor.epl_01a_target_1_active
sensor.epl_01_target_1_angle 	sensor.epl_01a_target_1_angle
sensor.epl_01_target_1_distance 	sensor.epl_01a_target_1_distance
sensor.epl_01_target_1_resolution 	sensor.epl_01a_target_1_resolution
sensor.epl_01_target_1_speed 	sensor.epl_01a_target_1_speed
sensor.epl_01_target_1_x 	sensor.epl_01a_target_1_x
sensor.epl_01_target_1_y 	sensor.epl_01a_target_1_y
binary_sensor.epl_01_target_2_active 	binary_sensor.epl_01a_target_2_active
sensor.epl_01_target_2_angle 	sensor.epl_01a_target_2_angle
sensor.epl_01_target_2_distance 	sensor.epl_01a_target_2_distance
sensor.epl_01_target_2_resolution 	sensor.epl_01a_target_2_resolution
sensor.epl_01_target_2_speed 	sensor.epl_01a_target_2_speed
sensor.epl_01_target_2_x 	sensor.epl_01a_target_2_x
sensor.epl_01_target_2_y 	sensor.epl_01a_target_2_y
binary_sensor.epl_01_target_3_active 	binary_sensor.epl_01a_target_3_active
sensor.epl_01_target_3_angle 	sensor.epl_01a_target_3_angle
sensor.epl_01_target_3_distance 	sensor.epl_01a_target_3_distance
sensor.epl_01_target_3_resolution 	sensor.epl_01a_target_3_resolution
sensor.epl_01_target_3_speed 	sensor.epl_01a_target_3_speed
sensor.epl_01_target_3_x 	sensor.epl_01a_target_3_x
sensor.epl_01_target_3_y 	sensor.epl_01a_target_3_y
number.epl_01_zone_1_begin_x 	number.epl_01a_zone_1_begin_x
number.epl_01_zone_1_begin_y 	number.epl_01a_zone_1_begin_y
number.epl_01_zone_1_end_x 	number.epl_01a_zone_1_end_x
number.epl_01_zone_1_end_y 	number.epl_01a_zone_1_end_y
binary_sensor.epl_01_zone_1_occupancy 	binary_sensor.epl_01a_zone_1_occupancy
number.epl_01_zone_1_occupancy_off_delay 	number.epl_01a_zone_1_occupancy_off_delay
sensor.epl_01_zone_1_target_count 	sensor.epl_01a_zone_1_target_count
number.epl_01_zone_2_begin_x 	number.epl_01a_zone_2_begin_x
number.epl_01_zone_2_begin_y 	number.epl_01a_zone_2_begin_y
number.epl_01_zone_2_end_x 	number.epl_01a_zone_2_end_x
number.epl_01_zone_2_end_y 	number.epl_01a_zone_2_end_y
binary_sensor.epl_01_zone_2_occupancy 	binary_sensor.epl_01a_zone_2_occupancy
number.epl_01_zone_2_occupancy_off_delay 	number.epl_01a_zone_2_occupancy_off_delay
sensor.epl_01_zone_2_target_count 	sensor.epl_01a_zone_2_target_count
number.epl_01_zone_3_begin_x 	number.epl_01a_zone_3_begin_x
number.epl_01_zone_3_begin_y 	number.epl_01a_zone_3_begin_y
number.epl_01_zone_3_end_x 	number.epl_01a_zone_3_end_x
number.epl_01_zone_3_end_y 	number.epl_01a_zone_3_end_y
binary_sensor.epl_01_zone_3_occupancy 	binary_sensor.epl_01a_zone_3_occupancy
number.epl_01_zone_3_occupancy_off_delay 	number.epl_01a_zone_3_occupancy_off_delay
sensor.epl_01_zone_3_target_count 	sensor.epl_01a_zone_3_target_count
number.epl_01_zone_4_begin_x 	number.epl_01a_zone_4_begin_x
number.epl_01_zone_4_begin_y 	number.epl_01a_zone_4_begin_y
number.epl_01_zone_4_end_x 	number.epl_01a_zone_4_end_x
number.epl_01_zone_4_end_y 	number.epl_01a_zone_4_end_y
binary_sensor.epl_01_zone_4_occupancy 	binary_sensor.epl_01a_zone_4_occupancy
number.epl_01_zone_4_occupancy_off_delay 	number.epl_01a_zone_4_occupancy_off_delay
sensor.epl_01_zone_4_target_count 	sensor.epl_01a_zone_4_target_count
update.epl_01_firmware 	update.epl_01a_firmware
device_tracker.epl_01 	device_tracker.epl_01a
sensor.epl_01_uptime 	sensor.epl_01a_uptime
```

EPL-02:
```
[... 33 lines cut ...]
binary_sensor.epl_02_zone_1_occupancy_2 | binary_sensor.epl_02a_zone_1_occupancy_2
number.epl_02_zone_1_occupancy_off_delay_2 | number.epl_02a_zone_1_occupancy_off_delay_2
sensor.epl_02_zone_1_target_count_2 | sensor.epl_02a_zone_1_target_count_2
number.epl_02_zone_2_begin_x_2 | number.epl_02a_zone_2_begin_x_2
number.epl_02_zone_2_begin_y_2 | number.epl_02a_zone_2_begin_y_2
number.epl_02_zone_2_end_x_2 | number.epl_02a_zone_2_end_x_2
number.epl_02_zone_2_end_y_2 | number.epl_02a_zone_2_end_y_2
binary_sensor.epl_02_zone_2_occupancy_2 | binary_sensor.epl_02a_zone_2_occupancy_2
number.epl_02_zone_2_occupancy_off_delay_2 | number.epl_02a_zone_2_occupancy_off_delay_2
sensor.epl_02_zone_2_target_count_2 | sensor.epl_02a_zone_2_target_count_2
number.epl_02_zone_3_begin_x_2 | number.epl_02a_zone_3_begin_x_2
number.epl_02_zone_3_begin_y_2 | number.epl_02a_zone_3_begin_y_2
number.epl_02_zone_3_end_x_2 | number.epl_02a_zone_3_end_x_2
number.epl_02_zone_3_end_y_2 | number.epl_02a_zone_3_end_y_2
binary_sensor.epl_02_zone_3_occupancy_2 | binary_sensor.epl_02a_zone_3_occupancy_2
number.epl_02_zone_3_occupancy_off_delay_2 | number.epl_02a_zone_3_occupancy_off_delay_2
sensor.epl_02_zone_3_target_count_2 | sensor.epl_02a_zone_3_target_count_2
number.epl_02_zone_4_begin_x_2 | number.epl_02a_zone_4_begin_x_2
number.epl_02_zone_4_begin_y_2 | number.epl_02a_zone_4_begin_y_2
number.epl_02_zone_4_end_x_2 | number.epl_02a_zone_4_end_x_2
number.epl_02_zone_4_end_y_2 | number.epl_02a_zone_4_end_y_2
binary_sensor.epl_02_zone_4_occupancy_2 | binary_sensor.epl_02a_zone_4_occupancy_2
number.epl_02_zone_4_occupancy_off_delay_2 | number.epl_02a_zone_4_occupancy_off_delay_2
sensor.epl_02_zone_4_target_count_2 | sensor.epl_02a_zone_4_target_count_2
```

EverythingSmartHome, 2024-12-09:
I think it's because they all have "_2" at the end of the entities. The way it works is that it's difficult to predict what all the entity names will be in the event that someone renamed the device, so it does rely on the entity name ending in "zone_1_begin_x". Though if you try and update to version 1.1.1, this might actually be resolved.

caretakerscurse, 2024-12-09:
I did try 1.1.1 with out a change.

I didn't even notice the *_2, so many entities have it I must have become blind to it.

It will probably take a while, but I'll rename them all and update here.

EverythingSmartHome, 2024-12-09:
Sure. It can certainly be improved and will be - it's very difficult to account for all the names people rename them too so at the moment we rely on the standard names

caretakerscurse, 2024-12-09:
Yup, that did it. 

I really wish that the bulk renaming tool would have removed the *_2... I had assumed that it would reset the names, but clearly I was wrong.

Thank you for the help. Sorry it ended up being the same issue everyone seems to have.

Take care out there.

closed 2024-12-09 by caretakerscurse