EverythingSmartHome/everything-presence-lite #263: Configurator not working

Ancalgon, 2024-12-07:
Hi guys, I have a problem.
Till few days ago the configurtion plugin was working like a charm, from yesterday (no updating anything) every time I start the UI to configure the zones, I get and error:

Error executing template: Failed to execute template

In the logs I can see:

`s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a Home Assistant Add-on.
 * Serving Flask app 'backend'
 * Debug mode: off
ERROR:root:Failed to execute template. Status Code: 400
ERROR:root:Response: {"message":"Error rendering template: Template output exceeded maximum size of 262144 characters"}`

The device is working, I had done a reset, I had tried also to create a docker on my server and connect to my home assistant but the error and the logs are identical

Other details:
ESPHome 2024.11.3
Device is not visible 
Error executing template: Failed to execute template

EverythingSmartHome, 2024-12-08:
Did you change the name of the device?

Ancalgon, 2024-12-09:
Yes the first time that I'd used the device, months ago. But the addon has worked for the first few days and then suddenly I've got the error. I had tried to to a backup of Home Assistant, roll-back to the day before I've installed the plugin and to install it fresh on the system, same problem from the first run. Of course was an experiment so I had rollforward to the newest backup.
But the sensor it self is working, something is strange because i had loosed all the zones data.

Ancalgon, 2024-12-15:
Nothing, I had reset another time my EP Lite, reinstalled the firmware, configured the wifi and exposed all the sensors. I had reinstalled ESPHome, remove any custom name but nothing. The configurator is not working and I really don't know how to solve the problem. The sensor itself is working

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a Home Assistant Add-on.
 * Serving Flask app 'backend'
 * Debug mode: off
ERROR:root:Failed to execute template. Status Code: 400
ERROR:root:Response: {"message":"Error rendering template: Template output exceeded maximum size of 262144 characters"}
ERROR:root:Failed to execute template. Status Code: 400
ERROR:root:Response: {"message":"Error rendering template: Template output exceeded maximum size of 262144 characters"}
ERROR:root:Failed to execute template. Status Code: 400
ERROR:root:Response: {"message":"Error rendering template: Template output exceeded maximum size of 262144 characters"}`

EverythingSmartHome, 2024-12-15:
Does it find the name of the device but doesn't work after that or does it not show the device in the drop down at all?

Ancalgon, 2024-12-15:
The configurator does not show any device. In home assistant the device hasn't any custom name, just the default one.

EverythingSmartHome, 2024-12-15:
Can you share the ESPHome diagnostics by going to the device page in home assistant, click the 3 dots and download ESPHome diagnostics

Ancalgon, 2024-12-15:
[config_entry-esphome-01JF54CER4BJ5PHFYSFJT4638B.json](https://github.com/user-attachments/files/18140671/config_entry-esphome-01JF54CER4BJ5PHFYSFJT4638B.json)
[home-assistant_esphome_2024-12-15T14-02-32.603Z.log](https://github.com/user-attachments/files/18140672/home-assistant_esphome_2024-12-15T14-02-32.603Z.log)

Ancalgon, 2025-01-16:
I have tried to install the configurator as Home Assistant addon, as VM and LXC in Proxmox and as docker on Unraid. The Error is always the same. Interesting I get the error even if I don't give the home assistant ip. It happened inside the Unraid docker. And when I've configured the docker with the IP of my home assistant, for a brief moment something happen only to refresh the page after a second and end in the same popup error

Log of the addon inside Home Assistant
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a Home Assistant Add-on.
 * Serving Flask app 'backend'
 * Debug mode: off
ERROR:root:Failed to execute template. Status Code: 400
ERROR:root:Response: {"message":"Error rendering template: Template output exceeded maximum size of 262144 characters"}`

SamJongenelen, 2025-03-11:
Mine worked initially. I did rename my device! Renaming it back did not help, but have not rebooted

Ancalgon, 2025-03-30:
I have three units (two never used), two servers and I had tried everything that I could possibly think about. Nothing worked. I had treid to rename the frist unit, to reset it multiple times, to use home assistant's esphome, to use docker, to use vm for esphome, nothing. Simply the plugin is not woriking. I had tried to create a new home assitant server only for the second EPLite unite (never used). It doesn't work eather. I had treid to test it in a friend setup with the same result. 
The plugin had worked only for two days, after that, it hadn't worked. 
So sad, without is only a motion sensor

EverythingSmartHome, 2025-03-31:
> I have three units (two never used), two servers and I had tried everything that I could possibly think about. Nothing worked. I had treid to rename the frist unit, to reset it multiple times, to [...]

Can you upload the diagnostics? Go to settings > devices > esphome > everything presence lite > 3 dots, download ESPHome diagnostics

Ancalgon, 2025-04-08:
[config_entry-esphome-01JQM5JV159VWTAAYE5KTEE2QG.json](https://github.com/user-attachments/files/19655426/config_entry-esphome-01JQM5JV159VWTAAYE5KTEE2QG.json)