EverythingSmartHome/everything-presence-lite #266: Factory reset but stuck in bootloop (task watchdog got triggered)

JosephTico, 2024-12-12:
So a few hours ago my EP Lite just decided to randomly stop connecting to Wi-Fi, after weeks of working without issues. No matter what I tried, it just didn't seem to work and the red light kept flashing forever. So I just decided to use [the factory reset option to update the firmware](https://everythingsmarthome.github.io/everything-presence-lite/updating.html), however now I have a separate issue, as it seems the page does correctly ask for the Wi-Fi SSID/password and it connects, however the moment I unplug the device and reconnect it, it enters into a bootloop.

Relevant part of the logs:

```
[I][wifi:617]: WiFi Connected!
[C][wifi:428]:   Local MAC: 08:A6:F7:4F:06:E4
[C][wifi:433]:   SSID: [redacted]
[C][wifi:436]:   IP Address: 192.168.1.21
[C][wifi:440]:   BSSID: [redacted]
[C][wifi:441]:   Hostname: 'everything-presence-lite-4f06e4'
[C][wifi:443]:   Signal strength: -35 dB ▂▄▆█
[C][wifi:447]:   Channel: 5
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.1.1
[C][wifi:450]:   DNS1: 192.168.1.5
[C][wifi:451]:   DNS2: 45.90.28.78
[C][api:026]: Setting up Home Assistant API server...
[D][button:010]: 'get_mmwave_firmware' Pressed.
[D][switch:012]: 'mmWave Configuration Mode' Turning ON.
[D][switch:055]: 'mmWave Configuration Mode': Sending state ON
[I][app:062]: setup() finished successfully!
E (31898) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[18:49:33]E (31898) task_wdt:  - loopTask (CPU 0)
[18:49:33]E (31898) task_wdt: Tasks currently running:
[18:49:33]E (31898) task_wdt: CPU 0: IDLE0
[18:49:33]E (31898) task_wdt: CPU 1: IDLE1
[18:49:33]E (31898) task_wdt: Aborting.
[18:49:33]
[18:49:33]abort() was called at PC 0x400fda6c on core 0
[18:49:33]
[18:49:33]
[18:49:33]Backtrace: 0x40082966:0x3ffb08a0 0x400896e5:0x3ffb08c0 0x4008f89e:0x3ffb08e0 0x400fda6c:0x3ffb0950 0x400832e9:0x3ffb0970 0x40172bdb:0x3ffba970 0x400fe1bd:0x3ffba990 0x4008aabc:0x3ffba9b0
[18:49:33]
[18:49:33]
[18:49:33]
[18:49:33]
[18:49:33]ELF file SHA256: cf9fb5d3f2929d0f
[18:49:33]
[18:49:33]Rebooting...
```

Full logs attached

[logs.txt](https://github.com/user-attachments/files/18104098/logs.txt)

JosephTico, 2024-12-12:
Update: I think I found the issue myself; the problem is that my local DNS1 server was down, and for some reason the ESP instead of falling back to DNS2 as expected was just giving out and rebooting.

EverythingSmartHome, 2024-12-12:
Yes that's it, for some reason ESPHome doesn't seem to use secondary DNS servers. Glad you figured it out

closed 2024-12-12 by EverythingSmartHome