EverythingSmartHome/everything-presence-lite #269: Zone Configurator zones not saving

michaelblight, 2024-12-15:
I have v1.1.1 of the configurator and when I select the device it shows one zone called "HA Zone 1" with a red outline. If I drag boxes they are called "User Zone 1" and "User Zone 2" (etc) and have a purple outline and the co-ords are shown below "User Zones:". When I hit save (with Enable Persistence or Disable Persistence showing) it says "Zones saved successfully!". When I click Ok to dismiss the dialog, the "User Zones:" section is now empty, and the "HA Zone 1" is now moved to where "User Zone 1" was and "User Zone 2" (etc) is gone. There's nothing in the log suggesting it's not working.

I can't see any other issue suggesting there's a problem like this. I'm running HAOS 2024.12.03 on a Proxmox VM and the EPL is on firmware v1.2.0.

In the Chrome console when I hit save I'm getting multiples of the following error:
```
       POST http://192.168.0.174:8123/api/hassio_ingress/rz...Hc/api/services/number/set_value 400 (Bad Request)
saveZoneToHA @ script.js:1430
saveZonesToHA @ script.js:1337
script.js:1414 
```

Normally I access HA locally thru NPM as "https://mydomain.example.com" and this gets the same error, but above I've used the direct IP address to cut out that as a cause of the error.

EverythingSmartHome, 2024-12-15:
It sounds like you have zones 2-4 disabled in Home Assistant and soothe addon is unable to save the zones to home Assistant. You need to enable all the zones you want to use by going into the device page in Home Assistant and selecting an entity for each zone (they may show up as hidden entities), clicking the cog and then enable. Repeat this for all 4 entities for a zone

michaelblight, 2024-12-15:
You are very right! Thanks. I didn't think I disabled the entities, but cannot vouch for myself.

closed 2024-12-15 by michaelblight