EverythingSmartHome/everything-presence-lite #270: ESPHome connection lost after initial configuration.

Pelphry17, 2024-12-19:
I have both EP1 an EPLite. Initial configuration via bluetooth worked perfectly. Added to ESPHome and everything worked fine. Until power was removed. On reboot, both connect to WiFi, but not to ESPHome. 

I've tried re-flashing several times, both with BLE and without. Devices are discovered in HA after re-flash, when I try to add them I get "Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.". 

I am able to ping the address of the devices.

Ghaldur, 2024-12-20:
Just to mention that I encounter the exact same problem, I contacted the support and we're actually trying to solve this (it is a known issue but no out-of-the-box solution exists right now.

Apparently the ESP seems to struggle to reach github (for updates) and this is why this error occurs. I'm using PiHole and added the EPL to a whitelist and it worked for some time but now I have the same problem again.

From the last e-mail i had from support, it says : 
"A common issue we are seeing with customers just now is that the primary DNS server they have configured is non functional. ESPHome only seems to make use of the primary DNS server and not the secondary."

I'll try that and see, good luck with that and I hope we get a fix soon !

Pelphry17, 2024-12-20:
Thank You for the information. I was finally able to get my EP1 to work. My issue was I had it on a network that is not allowed internet access. As soon as I switched, it worked. Have not tried to go back the other network yet, or tried the EPLite's.

safetymonkey, 2025-01-06:
+1. I have the exact same problem, although there is one interesting twist: When the ESP is connected to the PC I used to flash it via USB (I have reflashed several times trying to fix this), it connects to Home Assistant and everything works fine. When I connect it to a basic power source, it still shows up on the network but gives this error. Very mysterious.

EverythingSmartHome, 2025-01-14:
It's not really mysterious - The reason it works after initial flashing is because (my understanding of it) there is logic to that knows to not run the code to check for updates, because when you first flash, it is obviously not connected to WiFi. Then once you enter the WiFi details and power cycle, typically the mechanism to check for updates kicks in and thus you run into the bug.

Its not only DNS, it can be the other things mentioned in this thread too such as firewall rules intentionally blocking it, the absence of firewall rules to allow access (depending on the way you set your network up), DNS, IPS, IDS etc.

What does your network setup look like? Can you post your full startup logs to make sure it's the same issue?

Ghaldur, 2025-01-14:
Well my temporary solution was to disable entirely PiHole and setting back the external DNS (not PiHole but CloudFlare, Google etc) as the DNS server for my whole network. I’m not experienced enough with PiHole to find another workaround since I tried to put the Everything Presence Lite in a whitelist but it doesn’t work because PiHole is not blocking anything from the EPL, but my understanding is that if PiHole is the DNS, the « rerouting to the external DNS » doesn’t work for the EPL and since we can’t set a DNS server directly in the EPL config, it will only use the one given by the router.

Unfortunately I don’t have any other solution and we will have to wait for an update allowing us to setup an arbitrary DNS into the EPL OR one letting it use secondary DNS (provided by the router).

Until then we’re kinda stuck, but at least we know that it’s the DNS setup you have to look for in order to find a workaround, depending on your network setup.

EverythingSmartHome, 2025-01-14:
For reference, this is the PR that we are waiting to be added into ESPHome, which will fix the problem: https://github.com/esphome/esphome/pull/8018

EverythingSmartHome, 2025-02-12:
Yes it's now resolved

closed 2025-02-12 by EverythingSmartHome