EverythingSmartHome/everything-presence-lite #272: Change WiFi network

FuturamaPhill, 2024-12-22:
I recently setup a new SSID for IoT devices and couldn't find any information online on how to switch the Everything Presence Lite (EPL) to a different AP. In the end I just connected it to my computer and re-flashed it using the setup guide.

Thought I would log this issue(more of a request) to:
1. Update documentation to advise re-flashing in the only way to update WiFi network. (Maybe add into FAQ section)
2. Suggest a future feature if its possible to allow reconfiguring the WiFi network without having to re-flash.

neilmartin83, 2025-01-01:
It's possible to add the following to the config to enable a fallback access point with captive portal.

```
wifi:
  ...
  ap:

captive_portal:
```

So if you either run a factory reset (to forget saved Wi-Fi creds amongst other things), or change your Wi-Fi SSID credentials, it will give you an open SSID to connect to with captive portal you can use to enter new Wi-Fi creds/do an OTA update etc. Can save having to take them off a wall and plug into a computer etc.

More info:

https://esphome.io/components/wifi#access-point-mode
https://esphome.io/components/captive_portal

I am using this in my EPL config without the BT Proxy (got dedicated ESP32s for that and the extra overhead on the EPL causes Wi-Fi instability for me). I don't know if this will add significant overhead if you also use BT Proxy with your EPL - so would suggest testing it out.

neilmartin83, 2025-01-01:
Also should add for completeness - if you are using BLE Proxy with EPL - you also get Improv which works in a similar way but gets the Wi-Fi creds over BT. However, you need another BLE Proxy or adaptor connected to your Home Assistant server to leverage that. So IMO the fallback AP/captive portal is a better option because you just need a phone/tablet/computer with Wi-Fi to use it.

https://esphome.io/components/esp32_improv.html

neilmartin83, 2025-01-01:
Last follow up - digging around, I can see the EPL also has Improv over Serial enabled.

So to change Wi-Fi networks - you could also:

1. Plug into a USB port
2. Go to https://web.esphome.io/
3. Click CONNECT
4. Choose the serial interface for the EPL from the list
5. Click the ellipses in the right corner and Configure Wi-Fi
6. Click CHANGE WI-FI and enter the new details

EverythingSmartHome, 2025-01-15:
Actually there shouldn't be any need to reflash - if it fails to connect to WiFi, it will start BLE Improv to allow you to set WiFi up again. You no longer need a bluetooth proxy or any bluetooth device attached to Home Assistant, it can be done directly via the Home Assistant iOS or Android App and uses your phones Bluetooth.

So if you want to reconfigure the WiFi, head over to settings > devices and it should appear in the list to setup, allowing you to enter WiFi details.

Hope that helps

closed 2025-01-15 by EverythingSmartHome