EverythingSmartHome/everything-presence-lite #273: Not able to add multiple zones in the configurator

helinut, 2024-12-22:
I can draw multiple zones and they appear to show up properly.  User Zone 1, 2, 3 look great.  When I click save zones, they all disappear except for the first and it's renamed HA Zone 1.

I've tried in Firefox and Chrome if that could make a difference.  Neither worked properly.

EverythingSmartHome, 2024-12-22:
It sounds like you have zones 2-4 disabled in Home Assistant. If they are disabled then it won't be able to save them.

Go into the device page in HA and find the 4 entities for zone 2, click the entity, hit the cog and then enable. Repeat for all entities in all zones.

Then the add-on should be able to save them

helinut, 2024-12-22:
Ah ha!  Boy I was looking everywhere.  May want to add that to your how to page unless I missed it.  Thanks a bunch.

closed 2024-12-22 by helinut