EverythingSmartHome/everything-presence-lite #275: Bluetooth keeps turning off

rjbutler, 2024-12-26:
Using the standard everything_presence_lite_ha.yaml, I find the mmwave_bluetooth switch keeps turning off.  I’m assuming that setting is required for it to be used as a bluetooth proxy?  Or am I mistaken.  Any idea why it gets turned off periodically?

EverythingSmartHome, 2024-12-26:
If you flash Bluetooth proxy then it's always on with no way to turn it off.

The "mmWave Bluetooth" is the Bluetooth that's built into the LD2450 that's used for updating firmware. Recommend to leave it off unless your doing a firmware update

closed 2024-12-26 by EverythingSmartHome

rjbutler, 2024-12-26:
Ah, ok.  That makes sense.  I was sure when I first installed it that Bermuda BLE Triangulation didn’t see the EPL until I turned that on, but maybe it just took a while to refresh.  Thanks!