EverythingSmartHome/everything-presence-lite #278: Zones are not saving

GoBooth, 2024-12-31:
I just received my Everything Presence Lite, so far so good.
I’m downloaded the add-on to configure the zones.

doesn’t matter if I create 2, 3 or 4 zones, when I click on save, I get a pop-up that says it has saved succesfully, but then the GUI deletes all the zones I just created.

If I check the entities, the zones don’t exist in the esphome/everything lite.

What am I doing wrong?

I’m on Firmware: 1.2.0

[image]

GoBooth, 2024-12-31:
ok I figured it out.

You need to **manually** enabled ALL the disabled entities related to all other zones (like 20 things)
and then when you create the zones, or more specificly, when you save the zones, "persistant" needs to be turned on.

GoBooth, 2024-12-31:
ok I figured it out.

You need to manually enabled ALL the disabled entities related to all other zones (like 20 things)
and then when you create the zones, or more specificly, when you save the zones, "persistant" needs to be turned on.

closed 2024-12-31 by GoBooth

noisymime, 2025-02-02:
I think this needs to be made a bit more clear in either the Zones documentation or the Zone Configurator. The message saying it was saved is entirely wrong and very confusing trying to figure out what is happening. 

Ideally the Zone Configurator should enable the required zones when you hit save.