EverythingSmartHome/everything-presence-lite #279: Using EPL for Home/Away presence using BLE on my iPhone/Android?

peterbarbosa, 2025-01-01:
Hello,

Not an issue, but more of a question - is there any way for the EPL to be my primary sensor for home/away presence detection on my HA so I can have automations based on when no one is home? Currently I'm using ESPresence and it works fine, but would love to see if I can somehow minimize BLE sensors across my house.

Thanks.

m4tthall73, 2025-01-07:
Bit confused by the question. The EPL has BLE so it isn't reducing the number of sensors across your house. I am guessing you want to replace you other BLE sensor(s) with EPL? If so, then sure, it will simply be replacing like for like. Unless I am misunderstanding the question - quite possibly! :-)

TripelJ, 2025-01-07:
I just want to add my 2 cents :) and I can't speak for everyone. But most, like my self decide to run on the non BLE code as the Bluetooth and WiFi shares the same chip, there can be some issues running them together at the same time. It is true it is capable of being a proxy as well, but again I would add dedicated ESP's for that and just let the EPL be on WiFi as the good sensor it is. The small m5stack is good to use as Bluetooth proxies btw.

m4tthall73, 2025-01-07:
I have mine with Bluetooth Proxy, BLE and WiFi and it has been working without any issues. However, I might be just lucky, so don't take this comment as any guarantee :-)

I have also recently just acquired an m5stack device and will look to be setting up soon.

closed 2026-05-30 by EverythingSmartHome