EverythingSmartHome/everything-presence-lite #281: Can't setup everywhere presence lite  in homeassistant.

glennpierce, 2025-01-05:
I just got the new everywhere presence lite and powered it up.
My homeassistant install detected it and I clecked the icon. It asked for my wifi ssid and password with i entered.
That was accepted but after the redirect link for configuration failed.

I believe this is because I did not have my: in my configuration ?

Anyway after that failed I do not see the device or integration in my homeassistant. 
However, I can see the device / client in my unifi router so I do have its ip address.

I have tried t add manually by going devices and integrations adding esphome. It asks for the host and I write in the ip address for the sensor. I the get "Unknown error occured" 
 
Looking at the log on the commandline gives

https://pastebin.com/j9rrhRMe

My homeassistant is a docker install
image: lscr.io/linuxserver/homeassistant:latest   fully updated

At this point I am lost. Is there a reset button to try again ? 

Thanks

glennpierce, 2025-01-05:
The following works for adding manually. 
Interesting this was the host and not the docker container.

sudo sysctl -w net.ipv4.igmp_max_memberships=50
sudo sysctl -w net.ipv4.igmp_max_msf=30

glennpierce, 2025-01-05:
I see the visual zone editor add on does not support installations with docker.
Is there another way to install or access the visual zone area configurator ?
Thanks

EverythingSmartHome, 2025-01-05:
I'm not sure about the error your getting, that is more of a Home Assistant thing rather than anything EPL related. Sounds like you figured it out though

Zone Configurator does work on docker, it's here in the docs: https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/creating-zones.html#advanced-alternative-install-with-docker-no-supervisor

closed 2025-01-05 by EverythingSmartHome