EverythingSmartHome/everything-presence-lite #283: unable to find user guide for mmwave position.

ChrisLee7, 2025-01-07:
Hello,
Just received 2 new lite's but the writing on the circuit board doesn't match the pictures in the wiki document
https://everythingsmarthome.github.io/everything-presence-lite/

I want to swap in a LD2410b for better static detection but the picture in the wiki has a picture pointing to the second row pin holes that has text "LD2410" pointing to 5 holes. 
My board has "see user guide for mmwave position pointing to the same place but there are 6 holes...

Where is the guide that explains where to place the sensor

[image]

Also, whilst i'm asking: does the board automatically recognise the other sensor or do I have to do extra config?

Thanks!

EverythingSmartHome, 2025-01-07:
The docs need updating for sure around other sensors.

The sensor goes in the same slot (the bottom of the two), but make sure to line up the 5v pin on the LD2410C with the VCC pin printed on the board.

[image]
[image]

If you head back to the "updating" page in the user guide, you can install the firmware for the LD2410C there

ChrisLee7, 2025-01-07:
Legend, thanks :)

closed 2025-01-07 by ChrisLee7