EverythingSmartHome/everything-presence-lite #284: Missing target

KonradJanssen, 2025-01-08:
One of my sensors too often doesnot detect a target, that is defenitely in range. I increased occupancy off delays to 60 seconds, but that is just not enough. The target is relacksing on the couch, at about 4 meter from the sensor. Is there a way to improve this (other than using a longer delay)?
In the screenshot you can see target 1 turns off frequently. At 16:25 the sensor was put at another location (powered down and back up again), and was looking good for some 12 minutes, but then failed again.
[image]

EverythingSmartHome, 2025-01-15:
Sorry for the delay - a 4m distance will be a little challenging for very small movements, depending on the positioning and angle. Getting closer is one way to improve it but not always possible, or a longer delay as you mentioned. If static presence is a top priority, then changing the sensor to either the SEN0609 or the LD2410C will provide much better static presence results, but those don't support zones.

KonradJanssen, 2025-01-15:
Thanks for the explanation. For now I'm using a long delay (600 sec), but I may be looking at the suggested sensors later on. Of my 3 sensors with this one I'm not using zones, so it migt me a good solution.

closed 2025-01-15 by KonradJanssen