EverythingSmartHome/everything-presence-lite #289: Unable to delete a zone

steveof2620, 2025-01-19:
This relates to the Everything Presence Lite - Zone Configurator / the EP Lite.

When I right click on an exclusion zone, I am prompted with "Delete Exclusion Zone". I click OK, but the zone remains in place.

[image]

bh7h, 2025-02-19:
@EverythingSmartHome Same thing here. I cannot delete an exclusion zone! I can only recreate it. Can you help?

asazm, 2025-03-01:
I'm having the same issue, after updating to the newer version of the configurator. Won't let me delete exclusion zone, nor any other zones, i can over write the regular zones, to redo, but would like to be able to delete exclusion and existing regular zones.