EverythingSmartHome/everything-presence-lite #291: EP Lite tracking a target, but occupacy says "off"/no detection and target 1 says "not active"

ndalbjergs, 2025-01-23:
Still the problem also mentioned in issue #129

The firmware in the EP-L MW is up to date after I have used the radar app (and yes is also says that it is the right/newst version)

The EPL still has times, where it tracks an object (the distance goes up and down when moving), but occupacy is "not registred" and Target 1 active says "off"

This really frustrating, and I have fixed it in my bathroom by setting up a zigbee motion sensor (battery), and the greated a sensor group i HA with both EPL and the zigbee in it.
This helps on the WAF in the household..

But that should not be needed.

I have tried to reset the EPL via. having it connected to a computer and start all over.. But of course the HA installation remembers the unit, when it is adapted in to HA again.

Is it a possibillity to remove it completely from HA and reset it, and then adapt it all fresh in HA (If yes, then how?)

I have several EPL, and the all have this issue.

[image]

[image]
[image]
[image]

EverythingSmartHome, 2025-01-23:
We think we've discovered the cause of this just 2 days ago, if your able to join the discord I can give you a firmware to try with the fix which a few others have reported resolved the out of sync issue

manuelmrmorgado, 2025-01-23:
@EverythingSmartHome I have this exact issue, started a few days ago, on one out of 4 ESPs. How can I join the discord to get that update?
Thanks!

ndalbjergs, 2025-01-23:
> We think we've discovered the cause of this just 2 days ago, if your able to join the discord I can give you a firmware to try with the fix which a few others have reported resolved the out of sync issue

Sounds great..

Where do I find the discord info?

ndalbjergs, 2025-01-31:
> We think we've discovered the cause of this just 2 days ago, if your able to join the discord I can give you a firmware to try with the fix which a few others have reported resolved the out of sync issue

This solution seems to do the trick!

Therefor I will close this issue.

closed 2025-01-31 by ndalbjergs