EverythingSmartHome/everything-presence-lite #293: Target Count 0 but Zone Occupancy frozen at detected

manuelmrmorgado, 2025-01-29:
I'm having an issue that sometimes the zones are clear, with target count 0 (correct) but the zones detection stays frozen at detected until I reset the sensor.

EverythingSmartHome, 2025-01-30:
Can you try version 1.2.2 which should resolve this

manuelmrmorgado, 2025-01-30:
@EverythingSmartHome thanks! Will give it a try.

nilsi, 2025-02-09:
I got this as well, it says the firmware is up to date with version 2024.12.4

turboc1208, 2025-04-06:
Here is what I'm seeing.  The sensors related to Target 1, go to a state indicating there isn't anything in the room about 30s after motion stops.  Then after about 1m after the target data clears, ( I have my general occupancy delay and zone 1 target delay set to 1m each ), the occupancy setting goes to clear.  
I am on hardware 1.0.7
Firmware 1.2.2
ESPHome 2005.3.2

Any thoughts?

EverythingSmartHome, 2025-04-07:
> Here is what I'm seeing. The sensors related to Target 1, go to a state indicating there isn't anything in the room about 30s after motion stops. Then after about 1m after the target data clears, ( [...]

Sorry I'm not following what the issue is for you here? Could you let me know in more detail?

EverythingSmartHome, 2025-04-07:
> I got this as well, it says the firmware is up to date with version 2024.12.4. It seems like it started working again after a "hard reboot" by pulling the cable. Did not help to reboot in the devices home assistant settings

That is not the latest version, please update to the latest which should resolve that

turboc1208, 2025-04-07:
Shouldn’t the target zone activity and occupancy agree with each other?Sent from my iPhone

EverythingSmartHome, 2025-04-07:
There isn't a target zone, it's either Target or Zone but they are different things. Can you let me know which one has the issue? Sorry I'm still not quite clear, perhaps a screenshot would help illustrate the problem?

turboc1208, 2025-04-07:
In settings there is an occupancy sensor and 4 target sensors. Occupancy does not clear when all four target sensors Activity is off. How is occupancy determined and how do the various off delays work with respect to each other?Sent from my iPhone

EverythingSmartHome, 2025-04-08:
The "Global" occupancy has it's own configurable timeout that is separate to the individual zones occupancies. When a target clears, then it will wait for the occupancy delay period that you have configured (60s I think you said) before it moves from detected to clear