EverythingSmartHome/everything-presence-lite #296: Unavailable after today's update.

FarleighRed, 2025-01-30:
Installed the update and now device is unavailable, seems to be showing as a new device in ESP Home but when I try to connect it says the device lost connection.

EverythingSmartHome, 2025-01-30:
Do you have it added to the ESPHome add-on? (Not integration)

FarleighRed, 2025-01-30:
It shows in the ESPHome Builder if that is what you meanSent from my iPad

EverythingSmartHome, 2025-01-30:
Yes that's right. Which entity did you use to apply the update?

FarleighRed, 2025-01-30:
It came up as an update like everything else in home assistant saying an update was available si I pressed to update!Sent from my iPad

Moustakio, 2025-01-30:
Same issue, update through the notification you get from home assistant.. now all entities appear as unavailable. I am able to see the device in esp builder.

Please help.

FarleighRed, 2025-01-30:
Just pressed install in esp home builder on one of my devices and that seems to have worked, just try on the other device

EverythingSmartHome, 2025-01-30:
The entity you updated with is disabled by default to avoid this issue.

The issue occurs when you have added to ESPHome builder (not required) and then you use the "Everything Presence Lite Firmware" entity (hidden by default) to update. That's because ESPHome builder automatically adds an encryption key to your device which is used in Home Assistant, and updating directly from GitHub removes the encryption key.

The best way to proceed is to remove from ESPHome builder, remove from HA and re-add it without the key, or re-install the firmware from ESPHome builder to add the key back, then disable the entity again to ensure it doesn't update from GitHub again.

I will close this for now as it's a duplicate of https://github.com/EverythingSmartHome/everything-presence-lite/issues/186

But that should get you going again

closed 2025-01-30 by EverythingSmartHome