EverythingSmartHome/everything-presence-lite #297: Potential feature request:  When EPL is mounted upside-down, the zone configurator should offer a switch to invert the image

FrankJaco, 2025-02-02:
For aesthetic reasons, I have my EPL mounded behind a canvas painting which required me to have the sensor mounted upside-down. This of course makes everything swapped left-to-right on the zone configurator.  It is possible to create zones this way, but certainly more difficult. I would like to see an "invert image" on the zone configurator to accommodate upside-down EPL mounting.

fj

johnmhedge, 2025-03-16:
Yes please.

crhammock, 2025-03-17:
Not sure why we are all trying to install our EPL upside down this year but I second that request.  Preferable the option could be in ESP code rather than the zone configurator would basically need to flip the x begin and end co-ordinates.  This would fix the same issue I have in HA and the zone tracking card I am using.

Gaffers2277, 2025-04-06:
Yeah, i just started testing out the Everything Presence Lite - Zone Configurator and it took me a minute to realise that the view was flipped because my EPLite is upside down. Definitely makes configuring a bit of a headache. Would be nice to see a toggle to flip the view.

lckdairesearch, 2025-05-13:
The official kit uses a ball joint, so you can probably rotate the whole thing so that the mount sticks upward and the USB port points down, then you have to do some cable management to make sure it doesn't dangle.

[image]

johnmhedge, 2025-05-14:
It would be very much appreciated if it could do this in the software.

EverythingSmartHome, 2025-05-29:
This should now be added in https://github.com/EverythingSmartHome/everything-presence-lite/pull/334

It'd be good to get some testing on this one though it's a fairly simple change