EverythingSmartHome/everything-presence-lite #300: Impossible to track more than one target

bklaesener, 2025-02-08:
I can track one target without problems. As soon as another target is present, the sensor displays both targets for not more than a few seconds and than only ever shows one target at a time, jumping back and forth between the two every few seconds.

I tested this behavior in different rooms and with different target locations.

The firmwares (mmWave sensor and device) are up to date (2.04 and 1.22). All device entities are enabled in Home Assistant.

The same erratic behavior can be observed in the HLK app.

bklaesener, 2025-02-08:
I tested distances between 1 and 3 meters. Makes no difference.

andjo, 2025-02-08:
I just had the same issue. No matter the location of the two targets, one of them had to leave range for the other to show up.
"Target 2 Active" was never set in HA.

Resolved it by connecting with the HLKRadarTool app and enable the "Multiplayer" option that was off. Then both HLKRadarTool and HA could show two targets.

bklaesener, 2025-02-08:
Thank you, I had already seen that setting and turned "Multiplayer" on in the app, but it makes no difference for me, unfortunately. After two seconds (barely) of showing two targets the tracking starts alternating between them.

EverythingSmartHome, 2025-02-08:
Try factory resetting the mmWave sensor with the entity inside of HA

bklaesener, 2025-02-09:
That did it, thank you!

bklaesener, 2025-02-09:
Unfortunately the fix was only good for a short while.

I was able to repeat the following several times (both targets are stationary in reality throughout the test):

1. Reboot "module" (in sensor app) or ESP (from HA).
2. Two targets are tracked correctly for ca. 1:30 minutes after the reboot has completed.
3. The two targets suddenly move towards each other, trade places (!) and then move apart again.
4. After another 30 seconds have passed, only one target is tracked at a time, alternating between the two targets.

PilaScat, 2025-02-10:
I have the same problem

alexbreezy92, 2025-02-25:
Same issue. The sensor does not do a good job tracking multiple targets despite being advertised as such.

PilaScat, 2025-03-20:
News about this?

EverythingSmartHome, 2025-03-21:
> Unfortunately the fix was only good for a short while.
> 
> I was able to repeat the following several times (both targets are stationary in reality throughout the test):
> 
> 1. Reboot "module" (in sensor app) or ESP (from HA).
> 2. Two targets are tracked correctly for ca. 1:30 minutes after the reboot has completed.
> 3. The two targets suddenly move towards each other, trade places (!) and then move apart again.
> 4. After another 30 seconds have passed, only one target is tracked at a time, alternating between the two targets.
> 
> ***UPDATE after a few more hours of observation:
> 
> * Target 1 is consistently detected
> * Target 2 disappears every few minutes and comes back after 30 to 90 seconds
> * Both targets are well within the detection area

Sorry I missed this - so are you saying that both targets are stationary and not moving?

bklaesener, 2025-03-21:
In this case it was me and my wife sitting on opposite sides of a desk, so stationary but obviously not completely immobile. The sensor was placed about 2m from her (Target 2) and about 1m from me (Target 1). Direct lines of sight to each of us.

EverythingSmartHome, 2025-03-27:
> In this case it was me and my wife sitting on opposite sides of a desk, so stationary but obviously not completely immobile. The sensor was placed about 2m from her (Target 2) and about 1m from me (Target 1). Direct lines of sight to each of us.

How does it respond if you test while standing up and moving?

PilaScat, 2025-03-27:
> How does it respond if you test while standing up and moving?

The problem is there in every state, stationary or moving

EverythingSmartHome, 2025-03-27:
Please get in touch via email through the shop

PoshAlchemist, 2025-06-01:
Did you ever get a resolution to this as I also am experiencing the same issue, I'm sat on the sofa, it detects me. My wife walks detects her but soon as she sits down , the other side of the couch it stops detecting her until she leaves the room and comes back in

Dodge883, 2025-07-28:
Try to add this:

```
switch:
  - platform: template
    name: "Multi Target"
    id: multi_target_switch
    optimistic: true
    disabled_by_default: `false`
    entity_category: config
    turn_on_action:
      - switch.turn_on: mmwave_configuration
      - delay: 2s
      - uart.write:
          id: uart_bus
          data: [0xFD, 0xFC, 0xFB, 0xFA, 0x02, 0x00, 0x90, 0x00, 0x04, 0x03, 0x02, 0x01]  # Turn on Multi Target
      - delay: 2s
      - switch.turn_off: mmwave_configuration
    turn_off_action:
      - switch.turn_on: mmwave_configuration
      - delay: 2s
      - uart.write:
          id: uart_bus
          data: [0xFD, 0xFC, 0xFB, 0xFA, 0x02, 0x00, 0x80, 0x00, 0x04, 0x03, 0x02, 0x01]  # Turn off Multi Target
      - delay: 2s
      - switch.turn_off: mmwave_configuration
```