EverythingSmartHome/everything-presence-lite #304: Compile Error with ESPHome 2025.2.0  [BUG]

ArthFink, 2025-02-19:
Hey, 

I just updated to the new release of ESPHome (2025.02), and got this error when compiling / updating the presence-lite

```

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/everything-presence-lite-e7ae60.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/7547/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.5.1
INFO Compiling app...
Processing everything-presence-lite-e7ae60 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp.o
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/event_emitter/event_emitter.cpp.o
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/http_request/http_request.cpp.o
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/http_request/http_request_arduino.cpp.o
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp: In member function 'void esphome::bluetooth_proxy::BluetoothProxy::send_connections_free()':
src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp:480:12: error: 'class esphome::api::BluetoothConnectionsFreeResponse' has no member named 'allocated'
  480 |       call.allocated.push_back(connection->address_);
      |            ^~~~~~~~~
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/http_request/http_request_idf.cpp.o
Compiling .pioenvs/everything-presence-lite-e7ae60/src/esphome/components/http_request/ota/ota_http_request.cpp.o
*** [.pioenvs/everything-presence-lite-e7ae60/src/esphome/components/bluetooth_proxy/bluetooth_proxy.cpp.o] Error 1
========================= [FAILED] Took 10.02 seconds =========================
```

if you need any further logs, please let me know

brave0d, 2025-02-19:
I've created issue on ESPHome - https://github.com/esphome/issues/issues/6769 but been advised to create it here.
And here we go - @ArthFink already reported the same issue. Hope it can be solved. Thanks!

szhorvath, 2025-02-19:
Same error for me.

EverythingSmartHome, 2025-02-19:
I think it must be (could be wrong) an ESPHome bug. There is no breaking changes listed for Bluetooth Proxy in the release notes and no config that needs changing as far as I can see from the release notes

MenesesPT, 2025-02-19:
This commit (https://github.com/esphome/esphome/commit/dd18a219db7e4d7306c4cbd088baecfd146a811d) changed the api component adding the _allocated_ member that the compiler is giving an error about. This isn't present in the PR https://github.com/esphome/esphome/pull/7547
It only errors on the bluetooth versions because the bluetooth proxy is the one that's trying to access the _allocated_ member

MenesesPT, 2025-02-19:
I'm also in favour of waiting for at least a few time!

EverythingSmartHome, 2025-02-19:
Yes, be aware you could run into an issue where values get out of sync

iay, 2025-02-20:
I made a comment over on the PR. Maybe the author will update it to be compatible with the current ESPHome release (obviously that would need to happen before it would be merged anyway).

iay, 2025-02-20:
That PR has now had the changes in the base component for 2025.2 merged in. I tested it on my install and it's now recompiling and running properly under 2025.2.

I performed a "Clean Build Files" first; I don't know if that's required but it's probably safest.

Taomyn, 2025-02-20:
I confirm, I updated both my EPLs and they are both working fine

brave0d, 2025-02-23:
@ArthFink this is actually solved only temporarily, and if https://github.com/esphome/esphome/pull/7547 is not going to be merged into 2025.3 then we will have again the same issue in March.
The problem is that here we are using some unstable features and author of this unstable feature already gave up... https://github.com/esphome/esphome/pull/7547#issuecomment-2675027043

If we want to keep using this sensor we either need the above feature or look for other solution directly here ...

EverythingSmartHome, 2025-02-24:
Specifically for the bug which is for 2025.2, this is now resolved. If it happens in 2025.3 then we can re-visit then. There is a little movement on the linked PRs now which is hopefully promising.

closed 2025-02-24 by EverythingSmartHome