EverythingSmartHome/everything-presence-lite #313: Zone Configurator shows two devices with the same name/id and there is no activity on either one

aram535, 2025-03-20:
I have a single EPL connected, the drop down in the "Select" shows the same device twice and I select either a popup tells me "No relevant entities found for this device.".  There is also no activity in the target tracking information.  The device is obviously working and working to well as at night I have dozens and dozens of false positives.  How do I diagnose this?

EverythingSmartHome, 2025-03-23:
Did you rename the entities manually? If so that would be one reason as the add-on looks for the entity names in their default naming scheme (difficult to find entities that have been renamed as infinite possibilities)

aram535, 2025-03-23:
@EverythingSmartHome I did. Anything I can do to get the new entity name in there manually?

EverythingSmartHome, 2025-03-23:
Custom names aren't supported at the moment, I considered adding it but it would result in you having to manually find and add like 30 individual entities which isn't a great experience - I'm sure there must be a way to find custom names more smartly and hope it can be added in future but for now only the standard naming scheme is supported sorry

aram535, 2025-03-24:
That would be great, right now the EPL is not usable as it goes off way too often.

closed 2025-03-24 by aram535

EverythingSmartHome, 2025-03-24:
Not sure what you mean there and how that relates to this issue sorry? Could you elaborate?

aram535, 2025-03-24:
The story is that I'm trying to reduce the zone of detection - I have home assistant which is via automation triggering a light, at night the light begins to almost flicker on and off on a random basis. I have tried various values for the mmwave but it's either too sensitive or not sensitive enough.  I haven't found a happy medium.  I'm now hoping to be able to reduce the zone of detection just to the door of my office to see if that resolves the issue.