EverythingSmartHome/everything-presence-lite #325: EPS lite devices keep transmitting as a wifi network

SynoUser-NL, 2025-04-20:
Hi,

All of my 5 ESP lite devices are working properly within Home Assistant. In my laptops/phones they all keep appearing in the WIFI access point list.
This is obviously unwanted and should imho be unneccessary. 

How can I switch this off? And shouldn't be off by default?
Or am I doing something wrong here..

Thanks!
DenW

EverythingSmartHome, 2025-04-20:
There is no code that enables this - are you running your own firmware? If so, can you show the config you have?

EverythingSmartHome, 2025-04-22:
Do you have them added to the ESPHome builder addon? Assuming you do since you mentioned newer esphome versions - if so, can you share the config?

SynoUser-NL, 2025-04-22:
I see I've made a mistake, the WIFI networks I'm seeing are not ESP lite devices but something else entirely!
My apologies, and closing this thread.

closed 2025-04-22 by SynoUser-NL