EverythingSmartHome/everything-presence-lite #326: Update from within HA or ESPHome fails

jandebeule, 2025-04-21:
I cannot get the firmware update to work.
The compilation process fails every time with the following output:

e7adf0/src/esphome/components/esp32_ble_server/ble_server.cpp.o
Compiling .pioenvs/everything-presence-lite-e7adf0/src/esphome/components/esp32_ble_server/ble_server_automations.cpp.o
Compiling .pioenvs/everything-presence-lite-e7adf0/src/esphome/components/esp32_ble_server/ble_service.cpp.o
Compiling .pioenvs/everything-presence-lite-e7adf0/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o
Compiling .pioenvs/everything-presence-lite-e7adf0/src/esphome/components/esp32_improv/esp32_improv_component.cpp.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/everything-presence-lite-e7adf0/src/esphome/components/esp32_ble_server/ble_server.cpp.o] Error 1
========================= [FAILED] Took 149.16 seconds =========================

It's been failing every possible update since 2024.12.4 (that's the version it is currently running on)

EverythingSmartHome, 2025-04-22:
Have you tried clearing the build cache on the config?

jandebeule, 2025-04-22:
Yes, I did... Same problem

EverythingSmartHome, 2025-04-22:
Can you share the config you have?

jandebeule, 2025-04-22:
No need anymore as I found the cause of the issue: it was ESPHomeBuilder add-on that had difficulties compiling.
Setting the parameter 'default_compile_process_limit: 1' for the add-on, restarting it and then compiling did the trick.

Thanks for offering your help and apologies for the non-issue on your end

closed 2025-04-22 by jandebeule