EverythingSmartHome/everything-presence-lite #329: Everything Presence Lite - Zone Configurator 1.1.3 - no activity is shown

aram535, 2025-04-26:
I upgraded to the latest version of the configurator (v1.1.3) and the entities (EPL and One) do now show up, however, when I select the EPL no activity is shown in the zone. It's just a empty radar map.  I have not been able to get a sense of where I am to start including or excluding.

Target Tracking Information is showing all N/A. 

Presence detection is working, a little too well (presence in the next room over is also being picked up).

EverythingSmartHome, 2025-04-26:
The zone configurator is only for the Lite, since the One does not support tracking. If the Lite does not show any activity, did you rename entities from the stock values? The configurator needs the default names in order to work

aram535, 2025-04-26:
This is just for the "lite" unit, although both show up in the drop down.  I had initially renamed the units, but have since removed and readded them without renaming them.  Prior to 1.1.3 the devices would just show a 'no entities found on this device' popup.  Now the device is pickable from the drop down and loads fine, just no tracking info.