EverythingSmartHome/everything-presence-lite #337: Question - How to update firmware

sew3521, 2025-05-29:
I apologize for what is likely a stupid question but I was wondering the best way to update the firmware for the Everything Presence Lite.
I currently have one with CO2 sensor connected to HA and it is running great. Currently I am running version 1.2.3. I do not see an option to update the firmware inside of HA.

Is the best way to update by connecting to my computer using the USB cord and then running the manual setup steps here?
https://everythingsmarthome.github.io/everything-presence-lite/updating.html

Thanks for the help.

sew3521, 2025-05-29:
Thanks for the response. This is what I see

[image]

Later today I will hook it up to one of my computers to troubleshoot it more. I will post back here after I do that.

sew3521, 2025-05-29:
I ended up reflashing manually using USB and I am not running the latest version. The Firmware option still shows unavailable.

[image]

Inside of ESPHome I do not have it selected for Full Control, should I?

[image]
I didn't do it because of this message that came up when I clicked Take Control

[image]

xpenno255, 2025-05-29:
For info, I don't think you are supposed to do the device firmware updates from home assistant if you have it adopted in esphome. It breaks the device and needs to be manually reflashed from esphome via USB. 

Esphome versions are ok to update from the HA GUI.

From my understanding the device firmware entity is hidden by default because of this. 

If you are using esphome then you should get the latest device firmware if you simply reflash from the esphome add-on over WiFi.

MiLeX84, 2025-05-31:
The devices are located and pointed to a certain position, the zones are setup to match it, removing the device to update them via USB is kina shit imo. The device should support OTA like everything else. And I don't have a laptop and can run around with to do this either. If OTA is not supported, that's big bummer.

MiLeX84, 2025-06-03:
I cannot update the Firmware either the versions are not even shown. 

[image]

xpenno255, 2025-06-03:
> The devices are located and pointed to a certain position, the zones are setup to match it, removing the device to update them via USB is kina shit imo. The device should support OTA like everything [...]

OTA should work just fine unless something else has gone wrong. The usb flash method is only required if you've previously incorrectly flashed the device or something failed during the flash. Just remember that if you manage it in esphome then update it in esphome. If you manage it in home assistant only then the home assistant OTA should work.

MiLeX84, 2025-06-03:
[image]
[image]

Not sure what you want to see, these are some of the other details, but I also have all the Controls and Sensors sections available to setup everything. Is there something specific you're looking for?

rvanmaanen, 2025-06-03:
> I cannot update the Firmware either the versions are not even shown.
> 
> <img alt="Image" width="596" src="https://private-user-images.githubusercontent.com/59506850/450622165-5612eba3-e4f9-46b5-a1e0-bd946f018fc7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg5NDIyNjgsIm5iZiI6MTc0ODk0MTk2OCwicGF0aCI6Ii81OTUwNjg1MC80NTA2MjIxNjUtNTYxMmViYTMtZTRmOS00NmI1LWExZTAtYmQ5NDZmMDE4ZmM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA2MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNjAzVDA5MTI0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzMzExYjQzOTM2N2VkNjI5OGU1NWZhZmZkMmFmNWE2NjIzZDQyNmMxZGVhMzQ0NmJhOGJmMmVjMDQxODg1ODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gkfoLtLunZZAVT6Z9MibTDctRPlU8nDhBR9Fd_M9Dwo">

Check if your device still has a working wifi connection / reconfigure it, that temporarily fixed it for me: https://github.com/EverythingSmartHome/everything-presence-lite/issues/337#issuecomment-2920155033

rvanmaanen, 2025-06-04:
Hmm, while I do have all IoT devices in a separate vlan, it's the same vlan as HA is running in and there is definitely outgoing internet access possible from that entire network. So I guess I have a different problem with the same symptoms. Hopefully I can figure it out as well

tgoodwin321, 2025-07-22:
Also having this issue. Stuck on 1.2.3. It’s connected to wifi with internet access. Have deleted and re-added, power cycled, etc. but no changes. Curious if anyone found a solution and is there a way to change wifi networks short of starting over? Thanks.