EverythingSmartHome/everything-presence-lite #340: EPL doesn't seem to work anymore with HA

terribleJim, 2025-06-02:
Not sure if this is an issue with Home Assistant or something with the EPL.

EPL no longer interfaces with HomeAssistant automations. I did try the HLKRadarTool and it showed the firmware up to date (a 2023 version) and it showed my position in the cone of detection.

I did an online (over wifi) update to the EPLite and lost connectivity, so I connected to my computer and updated via your web instructions. That seemed to go well. In ESPHome, it shows as online. (I have homeassistant installed via docker, as well as ESPHome and the everything-presence-mmwave-configurator.) The logs in the configurator container show problems 
`curl: (6) Could not resolve host: supervisor
[02:44:50] ERROR: Something went wrong contacting the API`

More of the log below. It is in the same vlan as the HA installation.. so shouldn't be a firewall issue. And it was working up until a few weeks ago when something went awry. Perhaps my updating HA from a version of late last year to the latest killed something. The time coincides I think.. didn't notice exactly when it quit working but maybe around that time.

Maybe someone could offer a way to get it back working with HA - not sure it is the fault of the EPL.

Those logs from the configurator container (not debug):

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh
curl: (6) Could not resolve host: supervisor
[07:41:22] ERROR: Something went wrong contacting the API
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
curl: (6) Could not resolve host: supervisor
[07:41:22] ERROR: Something went wrong contacting the API
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a standalone docker container.
 * Serving Flask app 'backend' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
ERROR:root:Exception occurred while connecting to Home Assistant API: HTTPConnectionPool(host='192.168.60.70', port=8123): Max retries exceeded with url: /api/ (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8ed25f30>: Failed to establish a new connection: [Errno 111] Connection refused'))
s6-rc: info: service legacy-services: stopping
s6-svwait: fatal: supervisor died
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/00-banner.sh
curl: (6) Could not resolve host: supervisor
[02:44:50] ERROR: Something went wrong contacting the API
cont-init: info: /etc/cont-init.d/00-banner.sh exited 0
cont-init: info: running /etc/cont-init.d/01-log-level.sh
cont-init: info: /etc/cont-init.d/01-log-level.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun backend (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
ERROR:root:Running as a standalone docker container.
 * Serving Flask app 'backend' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off`
[... 61 lines cut ...]
[10:12:16][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:16][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:17][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:17][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:17][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:18][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=3.8lx
[10:12:18][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:18][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:19][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:19][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:20][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:20][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=4.0lx
[10:12:20][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:21][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:21][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:21][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:22][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=4.0lx
[10:12:22][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:22][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:23][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:23][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:24][W][LD2450:867]: Expected 30 bytes but received 150!
[10:12:24][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=4.0lx
[10:12:24][W][LD2450:867]: Expected 30 bytes but received 150!
`

MenesesPT, 2025-06-02:
> ```
> [10:12:01][W][LD2450:867]: Expected 30 bytes but received 150!
> [10:12:01][W][LD2450:867]: Expected 30 bytes but received 150!
> [10:12:02][W][LD2450:867]: Expected 30 bytes but received 150!
> ```

In my experience this is due to an incorrect baud rate, you can try to set the baud rate back to the default of 256000 and rebooting the ld2450 in the HLKRadarTool app.

terribleJim, 2025-06-02:
> In my experience this is due to an incorrect baud rate, you can try to set the baud rate back to the default of 256000 and rebooting the ld2450 in the HLKRadarTool app.

oh my goodness.. That worked just dandy! Back to normal - many thanks!! I would never have figured that one out.

closed 2025-06-02 by terribleJim