EverythingSmartHome/everything-presence-lite #341: Two EPL devices seem to have the same device ID.  I cannot add the second one?  How to fix?

DarcyDetlor, 2025-06-06:
I am getting this error message when I attempt to add a second Everything Presence Lite.  

[image]

It was working at one point, but then it dropped out and failed to connect.  I removed it from HA, but when I attempted to add it back, I received this message.  The other EPL is working correctly.  How can I change the device ID of the one I am attempting to add to HA?

Neither of the options solves the problem, because it removes the working EPL and replaces it.  I then have the same problem trying to add it back to HA. 

Thanks

EverythingSmartHome, 2025-06-07:
I'm guessing you are using some units which have old firmware? This was an issue in units from a while ago, roughly 9 months old? If you want to update the units over USB first before adding them to HA, should resolve the issue.

You can update here: https://docs.everythingsmart.io/s/products/doc/restore-or-update-your-epl-to-factory-settings-using-a-computer-orj7PBBOJZ

DarcyDetlor, 2025-06-08:
Thank you, this fixed the problem.  Is there any way to know when new firmware versions are released?

EverythingSmartHome, 2025-06-08:
Nice!

Yes, Home Assistant will notify you in the update section

closed 2025-06-08 by EverythingSmartHome