EverythingSmartHome/everything-presence-lite #355: Can't update EPL

w1ll1am23, 2025-09-01:
I have three EPLs all on 1.0.3 firmware. 

I am not able to update them at all. Going to https://everythingsmarthome.github.io/everything-presence-lite/updating.html there is no connect button, I've tried three different computers (Safari, Firefox, and Chrome) assuming there is a bug with the site?

Is there a documented way to manually update these from the terminal? I can't find anything that explains how to do that.

EverythingSmartHome, 2025-09-02:
Hello, that's an old link. Use this one: https://docs.everythingsmart.io/s/products/doc/restore-or-update-your-epl-to-factory-settings-using-a-computer-orj7PBBOJZ

Btw that old link does work, you just need to make a selection on the multiple choice.

w1ll1am23, 2025-09-02:
🤦 well now I feel dumb. Yeah the new link is working just fine the old link I clearly didn't read well enough and it didn't look like the first selection was actually selectable. Thanks for getting back to me, running the updates now.