EverythingSmartHome/everything-presence-lite #357: Angle shows wrong way in editor

Sawtaytoes, 2025-09-05:
I think the angle you set is opposite of what shows up in the UI leading to a lot of confusion. Knowing this now, I think I understand why my zones have been so messed up.

This is how I set my zone:
[image]

In the image, I have angled my sensor 45 degress to the left of the wall. Sadly, if that was the case, the excluded ceiling fan would be on the left, not the right.

The proper setting is -45 degrees when angling left.

[image]

Now that I have it this way, the ceiling fan still shows up on the right, but the angle in the image doesn't match how I rotated the sensor.

The way I see this image, the top is the back wall, the east side is the left, and the west side is the right.

Sawtaytoes, 2025-09-05:
It's sitting on the top of a table facing left.

Here's a very rough approximation of what it looks like:
[image]

EverythingSmartHome, 2025-09-05:
Sorry I don't understand the issue, for me personally this looks correct and all makes sense with the settings you have (the minus 45 degrees). You mentioned that the angle in the image doesn't match the angle in real life, but it seems to me it does. Sorry if I am missing something

closed 2026-05-08 by EverythingSmartHome