EverythingSmartHome/everything-presence-lite #358: Connecting Everything Presence Zone Configurator Docker to HA failed

JBS5, 2025-09-13:
I’m trying to run the Everything Presence Zone Configurator in a standalone Docker container using docker compose, following the documentation:
[Docker Compose example](https://docs.everythingsmart.io/s/products/doc/how-to-create-zones-2ApXUdPYAz#h-docker-compose-example)

I’ve created a new long-lived access token in Home Assistant, but the connection from the container still returns an error.
The Home Assistant URL is correct, as HA itself shows the following notification:

`Login attempt or request with invalid authentication from 172.18.0.9 (172.18.0.9). Requested URL: '/api/'. (python-requests/2.32.4)
`

Docker container logs:
```

2025-09-13T15:53:30.110026189Z ERROR:root:Cannot send to HA - WebSocket not connected
2025-09-13T15:53:30.110544190Z ERROR:root:Cannot send to HA - WebSocket not connected
2025-09-13T15:53:34.694784308Z ERROR:root:HA WebSocket error: [Errno -3] Try again
2025-09-13T15:53:34.695097325Z ERROR:websocket:[Errno -3] Try again - goodbye

```

What I’ve tried so far:

- Added websocket_api: to configuration.yaml (https://www.home-assistant.io/integrations/websocket_api/)
- Verified the token works via curl
- Environment variables in the container are correctly loaded

Has anyone encountered this issue or has any idea why the container still not able to connect to Home Assistant, despite a valid long-lived token?

JBS5, 2025-09-15:
Looks like creating zones using the 'Everything Presence Zone Configurator' container is working despite the errors in the container log and the offline/online/offline/online in the UI. 

So, maybe we can ignore the errors which are thrown?

Mi-Nigle, 2025-10-07:
I am having this issue.

The container logs are the same as yours @JBS5, and the connection status keeps switching from connected (for 5 seconds) and disconnected (2 seconds). I can see the EPL in the dropdown, and I can open its settings. Turning the LED on and off works, for example.

However I cannot see any live positioning. It's just a blank radar. All the zones show as entity disabled and hovering over any of them reveals a message _"This zone is disabled in home assistant. To enable it, go to Settings > Devices & services > Entities, search for the zone entities, and enable them."_ Zone 1 is definitely enabled in home assistant.

Also, I can draw out a new zone and when I select save zones, it does update the zone 1 values in home assistant. But with no live tracking it's not useful.

If it's relevant, this is my docker-compose.yaml (with redacted bits)
```yaml
  zones:
    ports:
      - 8099:8099
    image: everythingsmarthome/everything-presence-mmwave-configurator:latest
    container_name: everything-presence-mmwave-configurator
    environment:
      - HA_URL=http://my_ha_url:8123
      - HA_TOKEN=my_valid_token```

A screenshot if useful.
<img width="825" height="764" alt="Image" src="https://github.com/user-attachments/assets/ebe16a4b-64f8-4442-9b17-f32db4845f15" />

EverythingSmartHome, 2025-10-07:
Judging by the name of your device, looks like it's been renamed? If so, this may be the issue as it only supports devices with the default names (we need entity names we can predict what they will be so we can accurately read from them)

closed 2026-05-08 by EverythingSmartHome